pymtl/pymtl3 #48: Syntactic sugar for s.connect

jsn1993, 2019-06-08:
Having a gigantic block of 
```
s.connect( s.x1, s.y1 )
s.connect( s.x2, s.y2 )
...
```
seems annoying.
We are considering adding some syntactic sugar 
```
s.x1 //= s.y1 
s.x2 //= s.y2
...
```
to wrap around s.connect.

There are a few selections:
1. Assign-based ```|=, <<=, /=```. The good thing about assign-based is the implied directionality, but WHENEVER WE INTRODUCE A CONSTANT, it becomes really ambiguous and confusing:
```
s.x |= 1
s.y <<= 2
s.z /= 2
```
2. ```//```. This operator is perfect except that when we implement both floordiv and rfloordiv, 
```
123 // s.out
```
looks seriously like a C/verilog style comment ...

jsn1993, 2019-07-11:
Actually we might want to make connect a free function instead of s.connect()

cbatten, 2019-07-11:
I agree! Making `connect()` a free function helps make it easier for others to create their own connect functions like:

 - `connect_auto`
 - `connect_pt2pt`
 - `connect_grid`
 - `connect_chain`
 - `connect_ring`

etc ... and I think we still want the `//=` syntatic sugar with support for lambdas too:

```
 s.out //= s.in
 s.out //= 32
 s.out //= lambda: s.in + 32
```

jsn1993, 2019-07-23:
This is already implemented in #71

closed 2019-07-23 by jsn1993