pymtl/pymtl3 #65: Abandon python2

jsn1993, 2019-06-23:
We should move to Python 3 ASAP

Zac-HD, 2019-06-23:
OK, tips on how to do this: merge #52, then get the source code working on both 2 and 3, then remove Python 2.

(Currently, the source code is Python-2-only.  We run tests in CI for Python 3 by using `futurize`, which adds a dependency to emulate the Python 2 behaviour.)

1. Run [`modernize`](https://pypi.org/project/modernize/) on the code.  This adds a dependency on the very common `six` module, to support Python 3 behaviour on Python 2 (the opposite of `futurize`).  Commit the result - it is very useful later if the git history distinguishes between automated tools and any manual fixes to the results.
2. Run the tests on both Python2 and Python3, fix any problems, and commit the result.
3. Manual refactoring where e.g. `dict.values()` is cast to a list to match Python 2 behaviour, but you can iterate over it without casting.
4. When it's time to drop Python 2 support, run `pyupgrade --py3only`.  This can automatically remove the `six` compatibility layer, which is a major argument for `modernize` over `futurize`.

It will probably be easier to do steps one and two in stages, e.g. particular subpackages first.  Once that's done, you can do (3) and (4) in whatever order or whenever suits you.

jsn1993, 2019-06-24:
Hey Zac, I tried to run tests on python3 after I call modernize, but I got some error complaining about "can't concat str to bytes". I dig in a little bit and found that things like subprocess.check_output returns byte literal which wasn't a thing in python 2 ... Could you provide some idea on how to fix this? Thanks!

Zac-HD, 2019-06-24:
Call `check_output(..., universal_newlines=True)` to make it use text mode.  
(The argument has been renamed `text` in 3.7, but that doesn't help here).

jsn1993, 2019-06-24:
> Call `check_output(..., universal_newlines=True)` to make it use text mode on Python 3.
> 
> Confusing name for legacy reasons, but that's the trick.

Thanks Zac for this. Then I tried to do pyupgrade --py3-only, but it has no effect on the files. Am I doing something wrong?

jsn1993, 2019-06-25:
@Zac-HD I found that isort add "from __future__ ..." statements to some file in python3. As we abandon python2, do we still need them?

Zac-HD, 2019-06-25:
Nope; once you've dropped Python 2 support you can delete them, and [this line](https://github.com/cornell-brg/pymtl3/blob/master/.isort.cfg#L3) to stop `isort` putting them back.

Make sure you also add `python_requires=">=3.5",` to `setup.py`, to block installation on older versions of Python - that way someone installing from pypi will get the latest version that supports their version of Python.

ptpan, 2019-07-22:
I think it's time to migrate to Python 3 and I'm marking this as having high priority. If we would like to drop support for Python 2 then there is some further cleaning I can do before merging in #67.

jsn1993, 2019-07-23:
#67 has been merged

closed 2019-07-23 by jsn1993