pymtl/pymtl3 #72: Better handling of implicit `clk` and `reset` during translation

ptpan, 2019-07-19:
PyMTL 3 implicitly adds `clk` and `reset` ports to every component to ensure the framework has a reliable way to reset and tick components during simulation. Both `clk` and `reset` will be collected and translated as if they are ports added by users. However it's not safe to assume the desired clock and reset signal name are always `clk` and `reset`, especially when trying to integrate PyMTL-generated code with other RTL designs. 

One possible solution is to add flags to components indicating the exclusion of `clk` and/or `reset` signals. Under these flags, it becomes the user's responsibility to create and connect clock and reset signals correctly. Note that component simulation will still use the implicit `clk` and `reset`.

jsn1993, 2019-07-31:
Well in the port mapping dict can you map clk/reset to those clock/reset signals in the provided verilog source code?

ptpan, 2020-03-23:
Closing because we now support port map, `has_clk`, and `has_reset` flags.

closed 2020-03-23 by ptpan