pymtl/pymtl3 #75: Better names for translated components with parameters

ptpan, 2019-07-19:
The current PyMTL translation pass generates a pretty long name that includes the names and values of parameters. This naming scheme becomes awkward when a component has several parameters and when these parameters have long names. We even experienced verilator compilation error when the translated name of some component grows too long (> 100 characters?). Also there are situations where customized component names are useful (e.g. integrating PyMTL generated designs with larger projects whose interfaces are set).

We need (1) a way to make component names shorter (hashing?) and (2) to provide support for customized component names (a mapping between parameters and name?).

jsn1993, 2019-09-28:
Can we close this since #93 is already merged?

ptpan, 2019-11-12:
> Can we close this since #93 is already merged? @ptpan

The (2) part is still not done yet

jsn1993, 2020-03-23:
@ptpan is this done by our new placeholder/translate/import pass?

closed 2020-03-23 by ptpan