pymtl/pymtl3 #78: Integration with other Verilog-based projects

ptpan, 2019-07-25:
To better support other Verilog/SystemVerilog-based projects that use Verilator for simulation, it would be nice to have the following features:
1. Passing customized file list to Verilator (`-f` option)
2. Support for more Verilator options, especially those used to suppress warnings
3. Linking the shared library against other libraries
4. #73 
5. Instantiating a Verilog module with parameters
6. More advanced flags such as setting `rpath` during the actual C++ test bench compilation

ptpan, 2019-07-26:
@yo96 Any other things you would like to have for black-parrot testing?

yo96, 2019-07-26:
These look good to me. I think maybe we can organize all the flags read by import pass into an object called `settings` (like hypothesis does) so that we can set flags like this:
~~~
s.import_settings = ImportSettings(
  import = True,
  src_path = '../queue.v',
  list_path = '../flist.verilator',
)
~~~
This way it may be easier to manage all these flags (I'm sure there will be a ton of flags as we make the import pass more flexible) and it also saves name space. Anyway the general goal is to make the import as seamless as possible!

ptpan, 2019-07-28:
@yo96 Can you please remind me of why we wanted a special parameter `import_params` to the construction method of some placeholder to be imported?

yo96, 2019-07-28:
To compute the type for some ports

ptpan, 2019-08-09:
Closed as #80 has been merged into master

closed 2019-08-09 by ptpan