pymtl/pymtl3 #86: The use of `$` as separators during translation upsets some ASIC tools

ptpan, 2019-09-07:
We currently use `$` as a separator in the translation passes. Despite the fact that DC can handle signal names with `$`s pretty well, some ASIC tools fail subtly if they see signal names with `$`. For example, Synopsys vcat utility rejects a configuration file that mentions dollar-sign signals in the VCD file, without mentioning the use of `$`.

If `$` are generally not used in common Verilog source code, then we probably should consider switch to other characters (maybe `_`?) for the sake of better integration with other tools... Thoughts? @jsn1993

cbatten, 2019-09-07:
This is a great point. They key is to also avoid generating names that might conflict with names chosen by the designer. Maybe a double underscore since it is rare for a designer to use __?

ptpan, 2019-09-16:
Closing this issue because #87 has been merged

closed 2019-09-16 by ptpan