pymtl/pymtl3 #108: s.connect in tutorial slides

aliahari, 2019-11-15:
in tutorial slides (https://www.csl.cornell.edu/pymtl2019/pymtl-tutorial-ex01-isca2019.pdf) on pages like 25 and 29 you use s.connect() to connect two wires but it should be connect().

jsn1993, 2019-11-15:
Hi Aliahari, thanks for the interest in pymtl3! Yes, the tutorial slides only work with this tag https://github.com/cornell-brg/pymtl3/tree/0.2-isca2019. After the tutorial at FCRC, we have drastically improved the framework, but haven't got a chance to revamp the slides. You can either checkout this tag, or use the provided VM image to complete the tutorial with the slides, but the solution i.e. examples folder in the latest master should work. Sorry for the inconvenience.

aliahari, 2019-11-18:
Hi Shunning, 

thanks for the fast respond. I already found the code examples but you may just add a hint in the README of the main page that the syntax is changed compared to the tutorial time.

closed 2019-11-18 by aliahari