pymtl/pymtl3 #119: What if I want to simulate the off-chip data accesses ?

guoqinglei, 2019-12-16:
Is it necessary to add some memory simulation models such as DRAMSim2 if I want to do some simulation with memory access ?

jsn1993, 2019-12-16:
It should be doable to implement DRAMSIM2 as a black-box C++ module in pymtl3 standard library. We don't have plan to do so yet though. Another question is that the designer might want to use his own handshake or whatever interface, but we are only able to provide our own enable/ready interface implementation in the stdlib.

guoqinglei, 2019-12-17:
> It should be doable to implement DRAMSIM2 as a black-box C++ module in pymtl3 standard library. We don't have plan to do so yet though. Another question is that the designer might want to use his [...]

Thanks for you suggestions. I am new for pymtl3, could you tell me how to use your own enable/ready interface to access the DRAM memory ?

jsn1993, 2019-12-17:
> Thanks for you suggestions. I am new for pymtl3, could you tell me how to use your own enable/ready interface to access the DRAM memory ?

We currently don't have DRAMSIM2 support yet ... Need to investigate a little bit more.

yo96, 2020-01-29:
If my memory serves me correctly @ptpan and I imported DRAMSIM2 along with a cache from UW and tested it using pymtl3 when we were working with UW in Detroit last June. It's super hacky though.

closed 2023-12-06 by ptpan