pymtl/pymtl3 #123: Setting up renamed signals

JakeStevens, 2019-12-31:
I am writing a CL model and have a parameter that is set as an argument to my component. Based on that parameter, I would like to set some helper variables, like so:

```
def construct( s, x_major = True):
  if dim_major:
    # Rename to make things easier
    s.row_len   = s.num_y
    s.col_len   = s.num_x
    s.curr_row  = s.curr_y
    s.curr_col  = s.curr_x
  else:
    s.col_len   = s.num_y
    s.row_len   = s.num_x
    s.curr_col  = s.curr_y
    s.curr_row  = s.curr_x
```

where the items on the RHS are declared as Bits, and in the s.update block are updated using the +=1 increment, so that I can step through a grid. But, doing something like
```
s.curr_x += 1
```
actually creates a *new* Bits object and stores it in curr_x (i.e. it is not updating the value stored in the Bits object, since it is not modeling a register). And thus, something like `s.curr_col` would no longer point to the right object like I would like it to.

What is the best way to achieve this desired functionality using PyMTL3? I am currently just setting these helper variables in the s.update block, but I would prefer to have it outside the block in just the construct function.

I would also like to suggest the creation of a forum for PyMTL3, like those used for projects such as TVM. This would help prevent cluttering the github repo issues with questions like these, which are not related to the implementation but rather usage.

ptpan, 2019-12-31:
Thanks for your feedbacks Jake! Yes, the immutable feature of Python makes providing an alias for a supposedly mutable object (in your example `s.curr_x` and `s.curr_y`) pretty tricky. One solution without using update blocks is to declare these mutable objects as `Wire`s and set up name aliases through the `connect` API:

```python
class DUT( Component ):
  def construct( s, x_major = True ):
    s.curr_x = Wire( Bits32 )
    s.curr_y = Wire( Bits32 )

    s.curr_col = Wire( Bits32 )

    if x_major:
      connect( s.curr_col, s.curr_x )
    else:
      connect( s.curr_col, s.curr_y)

    @s.update_ff
    def upblk():
      if s.reset:
        s.curr_x <<= b32(0)
      else:
        s.curr_x <<= s.curr_x + b32(1)

    @s.update_ff
    def upblk_ff():
      print(s.curr_col)
```
and `s.curr_col` will correctly serve as an alias of `s.curr_x` during simulation 
```python
a = DUT()
a.elaborate()
a.apply( SimulationPass() )

a.sim_reset()
a.tick()           # 00000000
a.tick()           # 00000001
a.tick()           # 00000002
a.tick()           # 00000003
a.tick()           # 00000004
a.tick()           # 00000005
```
For CL modeling this can be cumbersome... If you want a pure-Python CL model that doesn't use RTL modeling primitives (wires and connections are mostly for RTL), I'd say setting up aliases through update blocks is the better option. 

Also thanks for the forum idea! The PyMTL team plans to make an official release in late January which includes documentation that hopefully will cover the most common use cases.

JakeStevens, 2020-01-02:
Thanks for the illustrative feedback. I look forward to the release later this month!

closed 2020-01-02 by JakeStevens