pymtl/pymtl3 #124: Conditional Elaboration

JakeStevens, 2020-01-08:
Is it possible in a CL model to have conditional elaboration? What I mean is the following:

I have a parametrized model with two modes-- for example, a CRC that operates either on one piece of data or an array. I'd like to have just one PyMTL CL model as follow:

```
class CRC( Component ):
  def construct(s, arr=True):
     < common defs>
     if arr:
       s.done = Bits1(0)
       s.length = InPort( Bits1 )
     <more stuff>

  @s.update
  def crc(s):
      <common stuff>
      if arr:
        if count == s.length:
          s.done = Bits1(1)
      <more stuff>
```

But when I attempt something like that, I receive an undefined error if arr is False, even though the undefined variables will never be reached. 

Is there an accepted PyMTL practice for this kind of setup?

Thanks!

ptpan, 2020-01-08:
Hi Jake,

The code you shown here seems to have some misalignment -- the update block function (`crc()` here) is supposed to be a free function inside `construct`. So your `crc` method should really be inside the `construct` method and defined without the `s` argument.

 I think Python may complain at the current `crc` method because `arr` is an unbound name in the scope `crc` is defined. Making `crc` a free function inside `construct` will solve that.

cbatten, 2020-01-08:
Right! Python's annoying indentation issues :) Here is maybe how it should look?

class CRC( Component ):
  def construct(s, arr=True):
     < common defs>
     if arr:
       s.done = Bits1(0)
       s.length = InPort( Bits1 )
     <more stuff>

  @s.update
  def crc(s):
      <common stuff>
      if arr:
        if count == s.length:
          s.done = Bits1(1)
      <more stuff>

JakeStevens, 2020-01-08:
Yes, @cbatten is right-- this is why I don't code in markdown! That is how my code looks, but I still have issues with it being undeclared. Specifically, if I am instantiating an object where arr == False. Then, I would receive an assertion error that:
AttributeError: 'FeatureFetcherCL' object has no attribute length

Basically, I believe that this is because the PyMTL backend doesn't "know" that the access to s.length is actually dead code when the component is instantiated with arr == False. I am not sure if there is a method for telling the backend this, or if there is a better way to achieve my goal.

Right now, I've just eliminated the if case in the constructor, but ideally-- particularly for RTL-- I would like to be able to not declare unused signals/interfaces/etc

cbatten, 2020-01-08:
Hmm ... maybe you could point us to a minimal code example and explain how to reproduce the issue. But I think what you are saying is your code works but it doesn't translate to Verilog because when we try and do the translation when arr = False then the translation pass complains that s.length doesn't exist ... which is technically correct and the right thing to do. We don't usually declare ports in an if statement like this ... so I am not sure if we have bumped against this before. Often we don't worry too much about unused ports because a good synthesis tool will not produce any overhead for unused ports ... here is a different idea. You could create a wire called `s.length_tmp` or something. Then if `arr` is true you simply create a port called `s.length` and connect the wire to the port but if `arr` is false you could connect the wire to a constant 0. Or wait ... because of dynamic typing and stuff you could actually just set `s.length = 0` if `arr` is false. That will probably work. If `arr` is true `length` is a port and if `arr` is false then `length` is a constant 0 ...

ptpan, 2020-01-08:
> Yes, @cbatten is right-- this is why I don't code in markdown! That is how my code looks, but I still have issues with it being undeclared. Specifically, if I am instantiating an object where arr == [...]

I think one solution would be to arrange your code like this

```python
  def construct(s, arr=True):
    <common defs>
    if arr:
      s.done = Bits1(0)
      s.length = InPort( Bits1 )
      @s.update
      def crc():
        <common stuff>
        if count == s.length:
          s.done = Bits1(1)
    @s.update
    def crc_not_arr():
       <common stuff>
       <more stuff from the original crc upblk>
    <more stuff>
```
Yes, PyMTL reports undefined variable because it does not do reachability analysis. The above arrangement guarantees `s.length` is defined when it is used in the `crc` upblk. However, this does introduce a little overhead of duplicating `<common stuff>` in `crc` -- you can maybe refactor that into a class method and call that method in the beginning of each `crc` update block.

JakeStevens, 2020-01-08:
I see. If there is no reachability analysis or cleaner way to do this, I think it would be better to simple have two separate components instead of trying to overload it in a clever way.. Thanks for the prompt responses!

closed 2020-01-08 by JakeStevens