pymtl/pymtl3 #126: Bit sizing in ops

JakeStevens, 2020-01-13:
I ran into some unexpected behavior as a result of the determination of the size the result of operations as specified [here](https://github.com/cornell-brg/pymtl3/blob/master/pymtl3/datatypes/PythonBits.py). For example, the behavior of add is defined as:

```
  def __add__( self, other ):
    try:    return Bits( max(self.nbits, other.nbits), int(self.value) + int(other) )
    except: return Bits( self.nbits, int(self.value) + int(other) )
```
However, I believe the correct behavior should be 
```
  def __add__( self, other ):
    try:    return Bits( max(self.nbits, other.nbits)+1, int(self.value) + int(other) )
    except: return Bits( self.nbits+1, int(self.value) + int(other) )
```
I believe that this definition matches the behavior of HDLs such as Verilog. For example,
```
module add(i_clk, a, b, out);

input	wire	[1:0]	a;
input	wire	[1:0]	b;
output	reg	[2:0]	out;

always @(posedge i_clk)
	out <= a + b;
```
would work as expected when a = 2 and b = 2.
However,
```
from pymtl3 import *
a = Bits2(2)
b = Bits2(2)
c = a + b
```
does not work as expected, as c results in Bits2(2), not Bits3(4).

Similarly, the bit width definition for multiply is incorrect, and should be the addition of the two bit widths (since unsigned multiplication is being defined as the default).

Finally, I believe that it would be beneficial to add in some sort of casting of constants to Bits, also to avoid unexpected results.

For example,
```
  def __add__( self, other ):
    try:    return Bits( max(self.nbits, other.nbits)+1, int(self.value) + int(other) )
    except:
        other = Bits32(other)
        return Bits(  max(self.nbits, other.nbits)+1, int(self.value) + int(other) )
```

Again, I believe that this matches the behavior of Verilog, where a constant without a defined bit_width is assumed to be 32 bits. Otherwise, you run into unexpected bugs, such as 
```
from pymtl3 import *
a = Bits2(3)
c = a + 3 # c now holds 2, instead of 4
```

cbatten, 2020-01-14:
We will have to give this some more thought. Implicit conversions of constants to Bits incurs non-trivial simulation performance overhead. That simple try/except you show above can incur multiple factors of slow down given how common simple bit-level operations are all ... and in the common case we are doing bit-level operations on bits. In PyMTL3 we must carefully balance traditional HW semantics with Python semantics and also balance simulation performance ... 

Regardless, we can certainly give this some more thought!

cbatten, 2020-01-14:
hmmm ... wait, I guess we already have that try/except block in our implementation ... we will have to think more about this. some of this also comes from PyMTL2 and might be worth rethinking ...

cbatten, 2020-01-14:
I should note that Verilog assuming constants are 32 bits is in itself a pretty big hack and can cause its own source of subtle bugs and frustrations. being explicit with the bit width of all constants is probably far better practice. we have a short hand that enables you to do this too `b32(1)` for a 32-bit constant one ...

cbatten, 2020-01-14:
The thing with the bitwidth+1 output for addition is also an easy work around. you simple zero extend the input operands appropriately before doing the addition. I think that is what we do ...

jsn1993, 2020-01-14:
https://github.com/dmlockhart/python-fixedbw-datatypes/blob/master/bitwidth_inference_comparison.rst
Here is the convention we follow.

cbatten, 2020-01-14:
you can also zero extend before doing multiplication. I remember that the PyMTL3 team actually spent quite a bit of time thinking about all of this. We decided to go with simple semantics with more things explicit than implicit. We do not allow implicit truncation which can cause lots of issues in Verilog ... so instead we use explicit extension before addition/multiplication to indicate the desired output bitwidth and we use explicit bitwidth constants. This adds a little syntactic overhead, but we think it results in cleaner and more explicit code ...

JakeStevens, 2020-01-15:
I see. It certainly makes sense to chose an explicit method. I think the issue is that these assumptions are not clear to the PyMTL user. I'm not sure there is a good way to provide a warning for the example where I am adding Bits2(3)+Bits2(3) and I want to know that the result is not Bits3(6). However, I think a warning could be produced for the case where it detects that I am using a constant with an unspecified bit width. For example,

  def __add__( self, other ):
    try:    return Bits( max(self.nbits, other.nbits), int(self.value) + int(other) )
    except:
        logger.debug('Using a constant with unspecified bit width')
        return Bits(  self.nbits, int(self.value) + int(other) )

That being said, I think providing a warning would only make sense if:
1) logging does not incur an overhead when not in debug mode. Basically, when developing a model, it would be important to see these warnings. However, when actually running simulations, the logger should not incur any overheads. I'm not actually sure if this is how the pythong logger module works or not.
2) The check/log is at the AST level or something, so that a line number of the unspecified constant can be provided.

cbatten, 2020-01-15:
Right! We are working on improving documentation. You can't really dynamically detect the situation where the input types don't match the output types because Python is a dynamically typed programming language and the = operator means you create a new result object (not assign a value to a previously allocated object). We do have some static type checking for RTL code during translation which I think should already detect this kind of bitdwidth mismatch.

JakeStevens, 2020-01-15:
Yes, definitely agree that Python being dynamically typed makes it hard (impossible?) to detect the input type/output type mismatch.

With that, I will close this issue.

closed 2020-01-15 by JakeStevens