pymtl/pymtl3 #135: Enforcing constraints across components

JakeStevens, 2020-01-24:
Is it possible to enforce constraints imposed on blocks **across** components? I am not quite sure how I would access such a block, since they are functions declared within the construct function. What I would like to be able to do is something like:

```
s.add_constraints(
    U(s.reader.AddrGen) < U(s.check_consistency),
    U(s.writer.Write) < U(s.check_consistency),
    U(s.check_consistency) < U(s.reader.Read)
)
```

where 'reader' and 'writer' are separate Components, which are instantiated inside of a MemoryController where I am trying to impose these constraints, and 'AddrGen', 'Write', and 'Read' are @s.update blocks inside those components

Thanks!
Jake

jsn1993, 2020-01-25:
Hi Jake, usually we have some data structures to carry the constraints for encapsulation purpose ... but if you really want to do it, check this out https://github.com/cornell-brg/pymtl3/blob/master/pymtl3/dsl/test/ComponentLevel1_test.py#L135

JakeStevens, 2020-01-26:
I see. So essentially, I can use the get_update_block function to achieve what I would like to do. But, the more "pymtl" way of doing it would be do generate some data structure so I can do something like call consistency_checker.rdy() within in the reader block, much like you would do with a queue from the stdlib.

closed 2020-02-17 by JakeStevens