pymtl/pymtl3 #139: No error or warning thrown when using `<<=` in a update block

yo96, 2020-01-29:
I just used `@s.update` instead of `@s.update_ff` for a sequential block and pymtl3 didn't complaint. When trying to translate it Verilator gives me this error:
```
E     Verilator output:
E     %Warning-COMBDLY: SwitchUnitMFlitRTL__a78c05a9be637c7c.sv:558: Delayed assignments (<=) in non-clocked (non flop or latch) block
E                                                                  : ... Suggest blocking assignments (=)
```
We should probably check this during elaboration? I think we check `@s.update_ff` but seems not `@s.update`

jsn1993, 2020-01-29:
it would require some deep hacking on the parsing process ... because it is possible for a CL model to use <<= in calculation.

yo96, 2020-01-29:
hmm if we have update once would that make this easier?

ptpan, 2023-12-06:
I can confirm PyMTL now throws the following error when using `<<=` instead an `@update` block:

```
E   pymtl3.dsl.errors.UpdateBlockWriteError:
E   In file /work/global/pp482/clean/pymtl3/pymtl3/stdlib/mem/ROM.py:47 in up_read_rom
E
E   47         s.rdata[i] <<= s.mem[ s.raddr[i] ]
E
E   ^^^ In update, only '@=' statements can assign value to signals, not <<=
E   (when constructing instance s of class "SequentialROM" in the hierarchy)
E
E   Suggestion: Line 47 Fix the '<<=' assignment with '@='
```