pymtl/pymtl3 #148: Response-Aware Memory Master

JakeStevens, 2020-02-24:
Hello,
    I was wondering if there is any module in the stdlib for a memory master that is aware of what type of queue is connected to the response port. Essentially, say I have:
```
s.sdram = MemMasterIfcCL( memreq_cls, memresp_cls )
s.sdramresp_q = NormalQueueCL(num_entries=3)( enq=s.sdram.resp )
```
I would like to have s.sdram.req.rdy() evaluate to true iff there *will be* room in s.sdramresp_q

This logic is relatively straightforward to do in my module, but I need to do this multiple times for different modules so I was wondering if there is any stdlib support for this. If not, and you think this could be broadly useful, I may be able to try to come up with a solution and push that out

jsn1993, 2020-02-24:
hmmm can you actually connect resp_q.rdy to sdram.req.rdy?

cbatten, 2020-02-24:
Hi Jake, We are trying out a new Google Groups mailing list for questions, suggestions, and discussion within the PyMTL user community. Can you maybe be our first (non-Cornell) member? Can you join the group and post your question there? Thanks!

cbatten, 2020-02-24:
oops ... here is the link: https://groups.google.com/forum/#!forum/pymtl-users

JakeStevens, 2020-02-24:
I have applied to join the group. After it is accepted, I will move this discussion to the forum and close this issue.

closed 2020-02-24 by JakeStevens