pymtl/pymtl3 #149: Bits call method

JakeStevens, 2020-02-24:
Is there a particular reason why the __call__ method is overwritten for the Bits datatype, as in:

```
def __call__( self ):
   return Bits( self.nbits )
```

It does not seem intuitive to me that calling an **instance** of a Bits object should provide another Bits object with the same number of bits, but equal to zero.

jsn1993, 2020-02-24:
oh, I think this code hasn't been touched for a while and I forgot why I wrote that lol. I don't think we actually use `__call__ ` in practice. I can remove that.

JakeStevens, 2020-02-24:
If there is no specific, reason, I think that may be good to remove; it can lead to confusing bugs if you're not expecting it

closed 2020-03-05 by jsn1993