pymtl/pymtl3 #191: Assuming other is not Bits?

JimJJewett, 2020-06-18:
https://github.com/pymtl/pymtl3/blob/a97f2a5a7d8d960907c8f7b4a9476632d0fc4707/pymtl3/datatypes/PythonBits.py#L218

Usually, it would use __add__ on the left operand instead, but not always.  For example, if self is actually an instance of a subclass of Bits, but other is an instance of regular old Bits.   The subclass might have some extra state to take care of, so its __radd__ is called instead of the parent's __add__.  (Recent optimizations may have changed this for cases when the subclass does not actually redefine __radd__.)

jsn1993, 2020-06-19:
Oh I think recently we already support Bits arithmetics with regular integer, so this TODO item can probably be removed. Are you referring to this?

JimJJewett, 2020-06-19:
It sounded like you wanted to micro-optimize it by assuming it was NOT a Bits.  (For example, just going straight to the int logic.)  Since it still could be bit-like ... I think the right solution is to just remove the TODO line, and *maybe* add a comment if you think you'll be tempted again later.

jsn1993, 2020-06-20:
> It sounded like you wanted to micro-optimize it by assuming it was NOT a Bits

We are trying to enforce a stronger restriction of arithmetics. If an operation is between Bits and non-Bits, we want to infer the non-Bits object as the same Bits type as the Bits object. If an operation is between two Bits objects, we want them to have the same bitwidth. I will remove the TODO.

closed 2020-06-20 by jsn1993