pymtl/pymtl3 #240: Verilator compile error

hmohajeri, 2023-02-05:
This operating system is ubuntu-22
The verilator 4.036 no longer compiles under ubuntu. It looks like some problem with Bison header. The verilator version that do compile under ubuntu however give errors during link with pymtl3. Is there a solution to help this problem? Here is the sample output of one of the runs from the class 

  C compiler output:
  In file included from IntMulScycleV1_noparam_v__ALL_pickled.cpp:1:
  IntMulScycleV1_noparam_v.cpp: In function ‘void trace(VIntMulScycleV1_noparam_t*, char*)’:
  IntMulScycleV1_noparam_v.cpp:304:23: error: ‘class VIntMulScycleV1_noparam’ has no member named ‘__VlSymsp’;
  did you mean ‘VIntMulScycleV1_noparam__Syms* const VIntMulScycleV1_noparam::vlSymsp’? (not accessible from this
  context)
    304 |   svSetScope( &model->__VlSymsp->__Vscope_IntMulScycleV1_noparam__v );
        |                       ^~~~~~~~~
  In file included from IntMulScycleV1_noparam_v.cpp:9,
                   from IntMulScycleV1_noparam_v__ALL_pickled.cpp:1:
  obj_dir_IntMulScycleV1_noparam/VIntMulScycleV1_noparam.h:21:42: note: declared private here
     21 |     VIntMulScycleV1_noparam__Syms* const vlSymsp;
        |

cbatten, 2023-02-05:
Hmmm ... no easy fix. @ptpan I think you had PyMTL3 working with the newest versions of Verilator right?

hmohajeri, 2023-02-10:
Hi @ptpan , any feedback on this item?

hmohajeri, 2023-12-07:
> Hi @hmohajeri <https://github.com/hmohajeri> -- sorry for getting back to
> you so late.. I've been woking on adding support for a recent version of
> Verilator (v5.016). The support is now complete in our dev branch and we
> are planning for a major revision of pymtl (#237
> <https://github.com/pymtl/pymtl3/pull/237>) in the near future so you can
> get that update from pypi. I apologize for this delay and hopefully this is
> helpful!

No worries and many thanks, I do understand that you must be very busy and
things take time.

all the best,