pymtl/pymtl3 #255: Driver with signals are shown as no drivers

KelvinChung2000, 2023-11-28:
The following example:

```
from pymtl3 import * 
from pymtl3.passes.backends.verilog import *

class a(Component):
    def construct(self):
        self.in_ = InPort(Bits32)
        self.out = OutPort(Bits32)

        @update_ff
        def test():
            self.out <<= self.in_

class b(Component):
    def construct(self):
        self.in_ = InPort(Bits32)
        self.out = OutPort(Bits32)

        @update_ff
        def test():
            self.out <<= self.in_

class top(Component):
    def construct(self):
        self.in_ = InPort(Bits32)

        self.a = a()
        self.b = b()
        
        self.a.in_ //= self.in_

        self.b.in_ //= self.a.out

if __name__ == "__main__":
    m = top()
    m.elaborate()
    m.set_metadata( VerilogTranslationPass.enable, True )
    m.apply(VerilogTranslationPass())
```

Yield the following error:

```
pymtl3.dsl.errors.NoWriterError: The following nets need drivers.
Net:
 - s.a.out
 - s.b.in_ 
```

This is not what I expected, as `a.out` is driven by `a.in_`  and `a.in_` is driven by `in_`.

ptpan, 2023-11-28:
Hi Kevin,

Thanks for filing this bug! I'm able to reproduce this locally and get the same error message as yours.

I also found that if you replace all `self` with `s` (e.g., `s.in_ = InPort(Bits32)`, `s.a.in_ //= s.in_`), this error goes away and produces the translation result. So it seems like the elaboration function assumes some the name of the current component needs to be `s`... I'm not exactly sure where that assumption happens though -- need to investigate further.

cbatten, 2023-11-28:
Thanks for the quick follow up @ptpan ! Odd that the elaboration pass makes an assumption the name of `self` vs `s` ... hopefully we can remote that restriction? @jsn1993 any thoughts on this?

KelvinChung2000, 2023-11-28:
After changing to `s`, the "supposed to have" errors pop up, like `@=` vs `=` in the `@update` block. Before I used `self`, they would just go through all the checks when I used `=`  in `@update` and generate some Verilog.

ptpan, 2023-11-28:
> After changing to `s`, the "supposed to have" errors pop up, like `@=` vs `=` in the `@update` block. Before I used `self`, they would just go through all the checks when I used `=` in `@update` and generate some Verilog.

Can you post the code you use and the error you saw with `s` instead of `self`? I'm not sure if I understand the "supposed to have" errors -- if you are doing RTL modeling and you should always use `@=` in `@update` blocks, and `<<=` in `@update_ff` blocks...

KelvinChung2000, 2023-11-28:
```
from pymtl3 import * 
from pymtl3.passes.backends.verilog import *

class a(Component):
    def construct(self):
        self.in_ = InPort(Bits32)
        self.out = OutPort(Bits32)

        @update
        def test():
            self.out <<= self.in_

if __name__ == "__main__":
    m = a()
    m.elaborate()
    m.set_metadata( VerilogTranslationPass.enable, True )
    m.apply(VerilogTranslationPass())
```
Something like this will run and generate the following Verilog.

```
//-------------------------------------------------------------------------
// a_noparam.v
//-------------------------------------------------------------------------
// This file is generated by PyMTL SystemVerilog translation pass.

// PyMTL Component a Definition
// At /home/kelvin/fullSystemDesign/test.py

module a_noparam
(
  input  logic [0:0] clk ,
  input  logic [31:0] in_ ,
  output logic [31:0] out ,
  input  logic [0:0] reset 
);

  // PyMTL Update Block Source
  // At /home/kelvin/fullSystemDesign/test.py:10
  // @update
  // def test():
  //     self.out <<= self.in_
  
  always_comb begin : test
    out <= in_;
  end

endmodule
```

ptpan, 2023-12-07:
Thanks for the reproducing example @KelvinChung2000! This has been resolved in #260 and merged into our dev branch. The change will be uploaded to pypi when we make the next major pymtl revision in the near future!

closed 2023-12-07 by ptpan