pymtl/pymtl3 #269: Incorrect VCD file format

KelvinChung2000, 2024-03-06:
I am trying to use a different waveform viewer ([surfer](https://gitlab.com/surfer-project/surfer)), which is not working. I have raised the issue to the `Surfer` team and found out there might be an encoding error in the VCD file for pymtl. 

This is the link to the [issue](https://gitlab.com/surfer-project/surfer/-/issues/202).

cbatten, 2024-03-06:
Hmm ... is it possible to provide a minimal test case that illustrates the issue along with what the desired VCD output should be? @jsn1993 any thoughts on this issue?

cbatten, 2024-03-08:
Thanks! We will try to find time to take a look ...

oscargus, 2025-07-06:
The problem is that you write `0b` before every bit-vector. This is not compliant with the VCD specification.

cbatten, 2025-07-06:
Right ... I don't think we really had time to look into this ... it works fine for us because we use GTKWave and Surfer ... I think the Surfer folks modified their VCD parser to accept what PyMTL3 generates :)

cbatten, 2025-08-22:
I think this is now fixed on the main branch:

- https://github.com/pymtl/pymtl3/pull/292

@KelvinChung2000 thanks for letting us know about this and @oscargus thanks for your help!