pymtl/pymtl3 #276: Using enum for design

KelvinChung2000, 2024-05-01:
I would like to use `Enum` to implement the design. 

Currently, I can write the statement like 
```
if (s.opcode == Opcode.LUI_OPCODE.value) | (s.opcode == Opcode.AUIPC_OPCODE.value):
    s.opcodeType @= InstType.U_TYPE.value
```

But this does not look very nice, especially when there are a lot of similar statements. If I would like to write like
```
if (s.opcode == Opcode.LUI_OPCODE) | (s.opcode == Opcode.AUIPC_OPCODE):
    s.opcodeType @= InstType.U_TYPE
```
Would this require significant changes to make this possible?

cbatten, 2024-05-01:
Can you provide a minimal example of what your code looks like? Is this purely for simulation or also for translation? Are you doing something like this:

```
from enum import Enum
class Foo(Enum):
  BAR = 1
  BAZ = 2
```

or this:

```
from enum import Enum
class Foo(Enum):
  BAR = Bits8(1)
  BAZ = Bits8(2)
```

So with regular Python if you want the enumerated constants to also behave as some other type you would inherit from a different base class like `IntEnum`. 

- https://docs.python.org/3/library/enum.html#enum.IntEnum

So with regular Python you cannot compare an enumerated constant using `Enum` to an `Int`:

```
from enum import Enum
class Foo(Enum):
  BAR = 1
  BAZ = 2

if   ( Foo.BAR == 1 ): print("BAR")
elif ( Foo.BAR == 2 ): print("BAZ")
else                 : print("???")
```

The above will print `???`. But you _can_ compare an enumerated constant using `EnumInt` to an `Int`:

```
from enum import IntEnum
class Foo(IntEnum):
  BAR = 1
  BAZ = 2

if   ( Foo.BAR == 1 ): print("BAR")
elif ( Foo.BAR == 2 ): print("BAZ")
else                 : print("???")
```

The above will print `BAR`. So one thing you might try is just using an `IntEnum` ... if you are defining your enumerated constants as `Int` that might just work? Here is an example I just tried:

```
from pymtl3 import *

from enum import IntEnum
class Foo(IntEnum):
  BAR = 1
  BAZ = 2

class DUT( Component ):
  def construct( s ):
    s.in_ = InPort  ( Bits8 )
    s.out = OutPort ( Bits8 )

    @update
    def block():
      if   ( s.in_ == Foo.BAR ): s.out @= Foo.BAR
      elif ( s.in_ == Foo.BAZ ): s.out @= Foo.BAZ
      else                     : s.out @= 0

input_values = [ 0, 1, 2, 3, 4 ]

dut = DUT()
dut.elaborate()
dut.apply( DefaultPassGroup() )

dut.sim_reset()
for input_value in input_values:

  dut.in_ @= input_value

  dut.sim_eval_combinational()

  print( f" cycle = {dut.sim_cycle_count()}: in = {dut.in_}, out = {dut.out}" )

  dut.sim_tick()
```
This is what I see:
```
 cycle = 3: in = 00, out = 00
 cycle = 4: in = 01, out = 01
 cycle = 5: in = 02, out = 02
 cycle = 6: in = 03, out = 00
 cycle = 7: in = 04, out = 00
```
Looks good? If you want to define your enumerated constants as `Bits` then I think you need to create a `BitsEnum` class in the spirit of `IntEnum`?

This is just for simulation ... for translation you would need to modify the translation pass to recognize Enums and generate the corresponding SystemVerilog code ....

Hope this helps!

KelvinChung2000, 2024-05-01:
For simulation and translation. The `IntEnum` is something I have never seen before. Using `IntEnum` works with the translation pass as well. Thanks.

closed 2024-05-01 by KelvinChung2000