pymtl/pymtl3 #280: Bit-replicate operator support

Aegithalos-caudatus, 2024-06-03:
Hi! Is there any direct support or equivalence for the bit-replicate operator in PyMTL? I am trying to implement some logic like this:
```
// Verilog
logic [2:0] A;
logic [NBITS : 0] B;

B = { NBITS{ A[1] }};
```
I searched for several related keywords in the repository but did not find useful information.

Also, I have tried the updates in [Improving concat #278](https://github.com/pymtl/pymtl3/pull/278). However, such generator expression is still rejected inside the update/update_ff block.

cbatten, 2024-06-04:
right ... we have the `sext` and `zext` operators ... if there are multiple bits to be replicated I think you would need to write a for loop in an update block?

Aegithalos-caudatus, 2024-06-04:
Thanks. I think a modern synthesizer should be able to yield the same result for these approaches. Though the for-loop version might be a bit harder to read ...