pymtl/pymtl3 #281: Bitwidth-mismatch check failed on customized bitstruct

Aegithalos-caudatus, 2024-07-31:
Hi, the [mk_bitstruct()](https://github.com/pymtl/pymtl3/blob/4b3bc183b14ffacd069f48dc5997b00045aea53f/pymtl3/datatypes/bitstructs.py#L733-L734) function allows customized naming for the generated bit struct class. 

Hence, when connecting such instances to normal BitsN instances, I think the bit-width check should be extended beyond the simple "_dsl.Type" match. Maybe add "_dsl.Type.nbits"?https://github.com/pymtl/pymtl3/blob/4b3bc183b14ffacd069f48dc5997b00045aea53f/pymtl3/dsl/ComponentLevel3.py#L256-L260

Debug code (o1 is BitStruct_49 and o2 is Bits49):
```    
def _connect_signal_signal( s, o1, o2 ):
    if not (o1._dsl.Type is o2._dsl.Type):
        raise InvalidConnectionError( f"Bitwidth mismatch {o1._dsl.Type.__name__} != {o2._dsl.Type.__name__}\n"
                                      f"--- Nbits info: o1 {o1._dsl.Type.nbits} != o2 {o2._dsl.Type.nbits}\n")
```
Error message:
```
E       pymtl3.dsl.errors.InvalidConnectionError: Bitwidth mismatch BitStruct_49 != Bits49
E       --- Nbits info: o1 49 != o2 49
E       Suggestion: make sure both sides of connection have matching bitwidth
```

cbatten, 2024-07-31:
Thanks for opening an issue! can you explain more? is your hope that connecting a `BitStruct_49` to a `Bits49` should not cause an error? If I remember correctly we did not want to make connecting a `BitStruct_49` to a `Bits49` work implicitly. We wanted the user to explicitly do something to do such a connection ... I think that is why we have the `connect_bits2bitstruct` function:
- https://github.com/pymtl/pymtl3/blob/4b3bc183b14ffacd069f48dc5997b00045aea53f/pymtl3/stdlib/connects/connect_bits2bitstruct.py#L47-L77
Or maybe you are asking something different?

Aegithalos-caudatus, 2024-07-31:
Thank you for the quick answer! I would like to have a direct assignment of flattened bits to/from the entire System Verilog packed struct. I was directly 
 using the "//=" without knowing that `connect_bits2bitstruct` function exists. Yet, as one signal is a "packed struct", my `connect_bits2bitstruct` attempt also fails because of 'A list of fields is not supported at the moment.'.

I agree that doing so is a kind of "knowing-exactly-what-you-are-doing".

cbatten, 2024-07-31:
right ... so I think we decided not to allow this favoring more explicit "casting" to avoid bugs ... you could always write your connect function to do what you want too

closed 2024-07-31 by Aegithalos-caudatus