pymtl/pymtl3 #283: pytest_cmdline_preparse deprecated

KelvinChung2000, 2024-10-30:
The `pytest_cmdline_preparse` hook is deprecated in version 7.0 and removed since version 8.0. This causes running simulation through pytest to result in an error.

```
INTERNALERROR>   File "/home/kelvin/fullSystemDesign/.venv/lib/python3.12/site-packages/pluggy/_manager.py", line 391, in check_pending
INTERNALERROR>     raise PluginValidationError(
INTERNALERROR> pluggy._manager.PluginValidationError: unknown hook 'pytest_cmdline_preparse' in plugin <module 'pytest_plugin.pytest_pymtl3' from '/home/kelvin/pymtl3/pytest_plugin/pytest_pymtl3.py'>
```

cbatten, 2024-10-30:
Yes. This is fixed on our pytest 4 dev branch. A PR on main is welcome if you would like to take a stab at fixing this.

KelvinChung2000, 2024-10-30:
I forgot to pull my 4.0 branch. Thanks.

closed 2024-10-30 by KelvinChung2000